Salisbury-University/robot-weed-killer #9: Robot drives forever on disconnect

qiugai, 2022-12-14:
Tick system is too intrusive. 👎

ZM009, 2022-12-15:
Might need to switch bluetooth modules as a __last resort__. 
HC-06 is only a **slave driver**, and AT mode is _not_ easily configured unlike the HC-05, HM-10, etc.

qiugai, 2022-12-15:
[image]

Potentially can tap into this LED header and use it to monitor the connection (depends on the age of our model).

ZM009, 2022-12-15:
> [image]
> 
> Potentially can tap into this LED header and use it to monitor the connection (depends on the age of our model).

Should be able to work, plastic sleeve needs to be removed to access pin 24 -- unless the STATE pin in our model is connected directly to LED. Can

ZM009, 2023-02-01:
Below code (sourced from [http://www.martyncurrey.com/hc-05-fs-040-state-pin/#:~:text=The%20STATE%20pin%20is%20LOW,value%20of%20the%20STATE%20pin.](url)), shows an example of using the HC-05 bluetooth modules state pin. This is a possible solution to detect connection/disconnection.

```
// BTconnected will = false when not connected and true when connected
boolean BTconnected = false;
 
// connect the STATE pin to Arduino pin D4
const byte BTpin = 4;
 
 
void setup() 
{
    // set the BTpin for input
    pinMode(BTpin, INPUT);   
 
    // start serial communication with the serial monitor on the host computer
    Serial.begin(9600);
    Serial.println("Arduino is ready");
    Serial.println("Connect the HC-05 to an Android device to continue");
 
    // wait until the HC-05 has made a connection
    while (!BTconnected)
    {
      if ( digitalRead(BTpin)==HIGH)  { BTconnected = true;};
    }
 
    Serial.println("HC-05 is now connected");
    Serial.println("");
 
    // Start serial communication with the bluetooth module
    // HC-05 default serial speed for communication mode is 9600 but can be different
    BTserial.begin(9600);  
}
```

ZM009, 2023-02-08:
Robot stops immediately on disconnect due to added code and hc-05 module switch