Salisbury-University/robot-weed-killer #12: Joystick Animation Glitching

ZM009, 2022-12-15:
## Expected Behavior

Joystick should run smoothly at ~30-40 frames per second when connected via bluetooth, while sending bt messages to the Arduino.

## Actual Behavior

When connected to bluetooth and sending navigation messages, the joystick animation stutters and glitches back and forth.

## Steps to Reproduce Behavior

* Launch app on smartphone
* Connect to the Arduino via the bluetooth menu
* Move joystick around while connected

### Possible Solution
##### Adding a Duration object to the Joystick View(?)
```
interval: Duration* ()
```

ZM009, 2023-02-15:
Will possibly fix at a later date; No solution at the moment.

closed 2023-02-15 by ZM009

qiugai, 2023-03-11:
Will not be fixed. More than likely phasing out joystick and focusing more on automated control first. (Slim chance to reopen later)