Salisbury-University/robot-weed-killer #19: Pixel Overflow on Screen Expansion

ZM009, 2023-02-01:
## Expected Behavior

When the webview is expanded to the entire screen of a phone, the buttons to control the robot manually should stay on top of the view, allowing the user to see the robot's camera view while still controlling its movement.

## Actual Behavior

When the screen is resized beyond certain dimensions (w x h), the joystick and other relevant buttons are pushed off the screen which causes a pixel overflow, and a while portion of the screen. 

------
#### Possible Solution

Row containing buttons may need to be expanded or redone from the ground up.

ZM009, 2023-02-08:
@evHitch i'll checkout webviewx

evHitch, 2023-02-15:
Implemented WebviewX & resolved the overflow issue. I'll be fixing the layout of the page next. 

[image]

ZM009, 2023-02-15:
Design needs work as it is rudimentary, otherwise approved.