Salisbury-University/robot-weed-killer #21: Redesign of Hardware

nickkrisulevicz, 2023-02-01:
Must research and decide on which new parts to include. This is necessary for two reasons. First, the parts must be known so their dimensions can be incorporated into the new designed frame. The new frame is being 3D printed and slots for each piece of equipment on the frame will be added in the 3D modeling phase. We cannot design the new frame nor print it until we choose the new parts. This is the first priority for the hardware design part of the project as it holds up progress on building the robot.

nickkrisulevicz, 2023-02-07:
After consulting with client, redesign of frame and body of robot will not take place within first milestone - before spring break. This issue will not be removed, but will not be addressed at this time

ZM009, 2023-02-09:
As of right now, can not be redesigned. Maybe will be updated after spring break and further feature developments.

closed 2023-02-09 by ZM009