Salisbury-University/robot-weed-killer #28: Goals Before Spring Break

nickkrisulevicz, 2023-02-07:
After consulting with client (Dr. Franchi), our team had a milestone laid out to be reached by spring break:

Priority one: The app needs to be functional and able to toggle between manual and autonomous operation. The app must be able to control the robot in its current state before major hardware upgrades are made. Initial autonomous operation must be achieved in an environment with no obstacles and square boundary. Additional pathfinding and navigation will be implemented later. For now, we need to focus on getting the robot to drive autonomously in a basic setting first.

Next milestone will follow after spring break once current one is done.

nickkrisulevicz, 2023-02-08:
UPDATE: This milestone was created in milestones, and it should not have been made in issues, this issue can be closed out

ZM009, 2023-02-08:
No longer needed

closed 2023-02-08 by ZM009