Salisbury-University/robot-weed-killer #33: Resistance - Finding Proper Ohms

nickkrisulevicz, 2023-02-13:
Dane will need to install resistors to the motors' hot wires to ensure that they do not burn up. 

Documentation has been provided by Zach to find the resistance needed given the voltage and amperage:

Below are current measurements listed for the "TT" motors.

To find voltage: V = I * R; (I = current, measured in Amps; R is resistance, measured in ohms)
To find resistance given input voltage and current: R = V / I

https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.adafruit.com%2Fproduct%2F3777%3Fgclid%3DCjwKCAiA3KefBhByEiwAi2LDHFsAweD2pzRRP57v9zoitfoc29EEQMsE5Yd-u_x1T0K0faNSF1b6yRoCa8MQAvD_BwE&data=05%7C01%7Cnkrisulevicz1%40gulls.salisbury.edu%7C042cc0c3570e49825c0e08db0dd3cca9%7C2472f1faf24f421badd7b01c4b49be07%7C0%7C0%7C638118975578246396%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=TxeWpMErZ%2FOrLRMt9dZ1kjutqLgnUYWcX4mdVDXh3ro%3D&reserved=0

Current priority: find out how much voltage and current the TT motors take: The measurements in the document are not official

ZM009, 2023-02-15:
Voltage was found to be a recommended 7.2 V per motor. 
Recommended current is 160mA (0.168 A)

Resistance found: $R=\frac{7.2}{0.168}$
$R=42.85\Omega\approx 43 \Omega$

ZM009, 2023-02-15:
Robot wouldn't drive with issues, will be revisited later.

closed 2023-02-15 by ZM009