Salisbury-University/robot-weed-killer #34: Inconsistent Access Point

ZM009, 2023-02-13:
## Expected Behavior

The ESP32 Camera should consistently broadcast the wifi network to different devices when nothing is connected.

## Actual Behavior

The Access Point of the camera is very inconsistent when broadcasting the wifi, and sometimes disconnects while displaying video feed, shutting off any view of what the robot sees.

### Potential Solution

* Possibly add more power
* Give camera a dedicated circuit

nickkrisulevicz, 2023-02-13:
A dedicated circuit was given to the camera, causing it to no longer draw insufficient power from the main circuit, this caused the camera to function completely and the access point now works consistently

ZM009, 2023-02-13:
### Testing

#### Multiple tests were completed by Dane and Nick with the video feed. Supervised by Zach and Garrett. Solution Approved. 👍

closed 2023-02-13 by nickkrisulevicz