Salisbury-University/robot-weed-killer #40: 💡 Create and Implement Basic Object Avoidance Algorithm

ZM009, 2023-02-15:
### Reference Issues

#38 #23

### Summary

#### Brief Algorithm Details:

* Stops motors when a few inches from an object
* Check around the object for other obstacles
* If obstacles are in all directions the robot will reverse and turn around to avoid all obstacles and return to path[^1]

[^1]: Based off basic object avoidance algorithms; will be wrapped in a conditional

### Basic Example

Written in c++/Arduino sketch, this feature will be used to avoid any and all obstacles detected by the ultrasonic sensor in a basic predetermined path.

### Drawbacks

#### Possible Impact:

* Only basic pathing
* Sonar may potentially fail
* Allows for further experimentation on sonar and possible coverage of multiple angles while driving

nickkrisulevicz, 2023-03-05:
Finite State Machine has been created from the simple flowchart algorithm. Now, the code needs to be created. Will be implemented in: main/Control/autonomous/auto.ino

ZM009, 2023-03-13:
Basic algorithm implemented -- in another issue, the algorithm will be heavily refined and possibly included with a mapping/path planning algorithm.

closed 2023-03-13 by ZM009