Salisbury-University/robot-weed-killer #41: Improvements and Reorganization of GitHub Wiki - Documentation

ZM009, 2023-02-21:
#### Started: 2/21/2023

## Goal
----
* Must add detailed instructions under each header to describe project overview, installation instructions, etc. within the main wiki page
* Start to record better documentation of Flutter extensions, dependencies, and Arduino modules.
* Consolidate documentation in one area for any and all libraries used.
* Record and store all sprint updates in separate section. 
* Create biweekly client update sections

nickkrisulevicz, 2023-04-16:
Will need to wait until all branches are merged before writing all READMEs or else chaos keeping track of all documentation could ensue.

nickkrisulevicz, 2023-04-16:
The Wiki is more important to anyone who views the repository than individual READMEs in each directory

closed 2023-05-17 by ZM009