Salisbury-University/robot-weed-killer #58: 🛠 Modify and Print Final Chassis Design

ZM009, 2023-04-04:
### Start Date

04/05/2023

### Reference Issues

#54 #51 

### Summary

The final design for the chassis will be printed and modified upon printing to add all necessary parts to the robot for further improvement of the autonomous algorithm.

### Basic Uses

## Improvements Explained
* Camera placement changed to prevent shake
* Move motor mounts for more freedom of design
* Used to further develop autonomous algorithm
* Integrated laser mount to prevent shake upon braking

nickkrisulevicz, 2023-04-05:
Designed new chassis iteration. Calculated weight distribution included in new directory located in sonar_patrol branch.

ZM009, 2023-04-07:
Chassis pieces printed in MakerLab

closed 2023-04-07 by ZM009