Salisbury-University/robot-weed-killer #59: 💡 Create and Implement Getters for Latitude and Longitude

ZM009, 2023-04-04:
### Start Date

04/04/2023

### Reference Issues

#57 

### Summary

Private getter functions would be created to utilize the latlng package's features in order to grab latitude and longitude measurements of custom points when the user holds an area of the map down.

### Basic Example

* Can create possible start and end points for robot path planning
* Could be used to track gps coordinates

### Drawbacks

Coordinates could have a bigger window of error; could require further refinement of the getters.

ZM009, 2023-04-10:
can use flutter_map's functionality to grab the latitude and longitude of a gps. Very high level implementation depending on the purpose.

nickkrisulevicz, 2023-04-16:
flutter_map works with the current GPS module just fine?

ZM009, 2023-04-16:
Yes. There's two libraries in flutter we can use to handle this: nmea, and latlong2. Latlong2 contains predefined types to contain the coordinates, while the nmea library is a parser that can be used to extract the given robot coordinates from the gps information sent by the arduino. These two can be tied in with flutter_map to reload the map's tiles and continually track the robot's location

ZM009, 2023-04-16:
After implementation there's a possibility of giving a robot a start and end point for path planning, but that may not be implemented due to time constraints.

ZM009, 2023-05-17:
No time left to learn nmea parsing

closed 2023-05-17 by ZM009