Salisbury-University/robot-weed-killer #62: 🐛 Screen Orientations Code Deprecated

ZM009, 2023-04-09:
### Description

When attempting to run the debug (or build) version(s) of the application, flutter catches and throws an error related to overriding a Navigator class function. This is due to deprecated code, and will need either a migrated fixed or a workaround.

### Reproduction steps

```bash
1. Change branch to dynamic_orientations
2. Connect device to laptop
3. Use flutter run --no-sound-null-safety
4. Check logs from console
```

### Logs

```bash
'NavigatorObserverWithOrientation.didPop' ('void Function(Route<dynamic>, Route<dynamic>)') isn't a valid override of 'NavigatorObserver.didPop' ('void Function(Route<dynamic>, Route<dynamic>?)').dart(invalid_override)
```

evHitch, 2023-04-09:
Found simple way to set preferred orientations on each respective page inside the initState() method.

evHitch, 2023-04-10:
respective page orientations set in initState() method.

closed 2023-04-10 by evHitch