Salisbury-University/robot-weed-killer #67: 🐛 Undefined Error for FlutterMap layers

ZM009, 2023-04-16:
### Description

When running the app in debug mode, the console logs an error relating to the layers list, marker options and tile layer options. 

### Reproduction steps

```bash
1. go to bt_controller:  robot-weed-killer/bt_controller
2. run command: ```flutter run --no-sound-null-safety``` (working on migration)
```

ZM009, 2023-04-16:
Fixed; found out method i was using was deprecated when reading flutter_map docs. pubspec auto updated to latest version, and i refactored accordingly.

take a look @nickkrisulevicz thanks :)

[image]

nickkrisulevicz, 2023-04-16:
Looks good. Let me know if there is anything I can do to help if any future problems arise

closed 2023-04-16 by nickkrisulevicz